tsdch-robotics/ftc5459 #6: Tracking bug for Base5459

nc4pk, 2016-02-09:
This is a tracking bug for implementing a "super-base class" for both autonomous and teleop modes.

The new base class extends `OpMode` instead of `LinearOpMode` to make teleop work correctly.

Things to do:
 - [ ] refactor initialization code out of teleop
 - [ ] change all autonomous programs to override `start()` instead of `runOpMode()`
  - *rationale*: `runOpMode()` only exists in LinearOpMode

Things to test:
 - [ ] our custom `initialization()` method changed to an override of `OpMode`'s `init()` method; does this change anything?
 - [ ] where should `waitForStart()` go? should it move outside the `init()` method?

nc4pk, 2016-02-09:
autonomous programs override `loop()` instead of `start()`

nc4pk, 2016-02-09:
`waitForStart() doesn't exist in `OpMode` so this method call was removed

tsdch-robotics, 2016-02-13:
- [ ] we need to override the `stop()` method and stop all drive motors

closed 2016-02-13 by tsdch-robotics